ZSWatch/ZSWatch #9: How to build the app?

ShadowThree, 2023-02-20:
Hello, I am very interested in this program, but I'm new for zephyr. Now I already install the dev env at of zephyr in ubuntu22.04 according to `https://docs.zephyrproject.org/latest/develop/getting_started/index.html`. But I can't find the way to build the project, could you add some short description for How to build the app in README? 
Thanks again for you awesome project.

jakkra, 2023-02-20:
Hi, I agree this should be there and will be soon. Actually I'm right now using an event library only present in the nordic nRF Connect fork of Zephyr (it's todo to make it build with stock Zephyr). But the nice thing with the nRF connect fork is that Nordic provide a simple tool to install Zephyr and the toolchain.

So to build:
- follow https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/getting_started/assistant.html#id8
- cd into the app folder
- follow https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/getting_started/programming.html
- west build -b smart_watch_anna_nrf52833

Let me know if this made it compile for you. I can write some better instructions tonight.

ShadowThree, 2023-02-20:
Hi, at last I built your app successfully. I got the below output(it seems the both of flash and ram are ran out of.):
[image]

Just recording for anyone who has the same issue:
1. Install [zephyr DEV ENV](https://docs.zephyrproject.org/latest/develop/getting_started/index.html);
2. Install [nRF connect SDK](https://developer.nordicsemi.com/nRF_Connect_SDK/doc/latest/nrf/getting_started/installing.html#build-environment-cli) by manually (Just do the steps that NOT including in `1`);
3. Copy the `app` folder in this program to `ncs/zephyr/app`;
4. Copy the `app/boards/arm/smart_watch_anna_nrf52833` folder in this project to `ncs/zephyr/boards/arm/smart_watch_anna_nrf52833`;
5. Change path to `ncs/zephyr/app`, then run `west build -p auto -b smart_watch_anna_nrf52833 .`;

That's all, good luck！

jakkra, 2023-02-20:
@ShadowThree that's great! 
Your step 1. should not be needed, both SDK and toolchain should come in 2.
Also should not need to copy anything, everything should work from within the app folder in the porject.
Boards folder should be picked upp automatically also.
I always build from the VSCode plugin, I'll make sure to have instructions for building with command line also.

About the RAM and flash, yes it's completely full :p But for v2 I'm using nRF5340, I just yesterday compiled the SW for it and now the flash usage is 41% and RAM likr 30% 🎆

ShadowThree, 2023-02-21:
@jakkra
Thanks for you goog job. I build the project by command line, where seems the VScode can't be used.
BTW, I can't wait to see your ZSWatchV2, haha.

JPHutchins, 2023-02-26:
I cannot successfully compile.  Is this built with Zephyr or NRF "Zephyr" 😄?  If NRF, WHICH NRF Zephyr lol?

JPHutchins, 2023-02-26:
Saw that you mentioned NCS 2.2.0 so I did get a build going!  Here are steps for Windows (I'll annotate for Linux/Mac compatibility but it's untested for now).

* Use Nodic's "Toolchain Manager" to install nRF Connect SDK v2.2.0
* cd to the `<root of this repo>`/app`
* Tell West to use the Nordic Zephyr is: `$env:ZEPHYR_BASE = "C:\ncs\v2.2.0\zephyr"` (Windows) `export ZEPHYR_BASE=<your NCS Zephyr location>` (Linux)
* Create a venv (you'll want to add it to .gitignore): `python3 -m venv venv` - technically you could put it anywhere (like the ncs folder, for use across multiple projects)
* Activate the venv: `. venv/scripts/activate` (Windows) `. venv/bin/activate` (Linux)
* Install dependencies: `pip install -r C:\ncs\v2.2.0\zephyr\scripts\requirements.txt` (update to your ncs install location)
* `west build -b smart_watch_anna_nrf52833`

You'd want to capture the two steps of "set ZEPHYR_BASE" and "activate venv" in an environment script for your environment.

This feels like less friction than I've encountered with Nordic's "tools" in the past 🙄 ... YMMV

jakkra, 2023-02-26:
Hi good that you got it building. If you use the open the command line from within the nRF Toolchain manager you can even skip your steps 2-5 ;) 
[image]

Or use their VSCode plugin which also only require like 3 steps in total. I'll add building instructions.

JPHutchins, 2023-02-27:
Cool!  FYI, I am using the build as a unit test here: https://github.com/JPHutchins/avstack/tree/main/tests/fixtures/ZSWatch

closed 2023-03-13 by jakkra