ZSWatch/ZSWatch #18: button

DREAM669, 2023-04-17:
I have been researching the structure of your first version recently, but I cannot understand your structure in the button section. How do you specifically implement the button function? I also noticed that you have a plastic part in the physical part of the button. Is this what you bought? Can you send me a purchase link to take a look? Thank you very much!

jakkra, 2023-04-17:
Hi, yes buttons I experimented around with a bit and there may be multiple variants in the CAD model, the one I ended up using was this one, however it's not what I want in V2, will come up with something better. It's two parts glued together.

I'm not sure how I should solve this, would be nice with a part I could buy instead of printing it.
Or perhaps something I could print in place that would then snap off, not a bad idea...

[image]

DREAM669, 2023-04-18:
Okay, I understand what you mean. You mean that the buttons for the first version were printed in 3D by you, right? I used to think you were buying watch parts, but I searched many stores but couldn't find them.

jakkra, 2023-04-18:
Yeah 3D printed all.

closed 2023-04-18 by jakkra