ZSWatch/ZSWatch #19: Is there a glitch when LDO cut-off by the voltage supervisor?

sam0910, 2023-05-19:
It is a Great Project :)
I'd like to try, but I have question about the battery supervior.

When the battery goes down to 3.6V -> NCV(supervisor) RESET_OUTPUT to low ->
Disable LDO(CE), Turn off MCU.

For charing battery,
VBUS connected -> charge battery -> if +BATT>3.6 turn LDO on by NCV

what if LCD BL and MCU drain more than charged capacity(might be low current VBUS), 
is there no glitch, like turn on and off in a loop?

jakkra, 2023-05-19:
Hi, thanks for the consern, I did not really think of this. 

So let's see:
First off USB should deliver at least 100mA per specification if I understand correctly. 
Worst case power consumtion of the PCB should be less than 100mA always.
- Display backlight maxes at 60mA + <5mA for the actual display.
- MCU worst case around ~15-20 mA (full radio TX and 100% CPU)
- All sensors are quite low power, so 20mA should cover for those and the rest of the components.
- If heart rate sensor is mounted however, this one if configured wrongly can consume a bit of power, so maybe as a precaution this should always be disabled when charging.

But assuming we don't charge with 100mA, which IMO is not normal use, the most important thing is that nothing breaks.

So what will happen:
The watch may turns on-off repeatedly, this is not an issue, nothing will break. However draining the battery is, BUT the battery I use, and many others have built in under-discharge-protection (althought often at too low voltage for the best battery health, hence I have this extra (battery supervisor) on the PCB), this internal battery protection should trigger and battery will not be harmed, unless leaving the watch like this for a long time.

So I think it should be fine, what do you think @sam0910?

sam0910, 2023-05-20:
It will be fine for me, and I think there will be no breaks at all.
it understood there is a small chance for glitching, 

I think this is most reliable project among all other open source project.
First of all, they don't care about battery life and consumptions.
I found most of them using $0.1 IC(TP4065) for battery charging.

LDO drains only from the battery that makes charging slow, 
I am just a web developer and asking your good opinion,
so what about adding some switcher like below? 

 
[image]

jakkra, 2023-05-21:
Thanks, I'll take this into consideration when doing next major version, I'm going to replace the charger and regulator with this one https://www.nordicsemi.com/products/Npm1100 so I'll see what more is needed than that.

sam0910, 2023-05-22:
That is great! it seems it's got LDO.