ZSWatch/ZSWatch #20: CAD

DREAM669, 2023-05-31:
From the picture, it can be seen that you have already made the watch case, haven't you? How beautiful! Can you share the CAD files and take a look? I am very curious about how the button part is resolved. Thank you.

jakkra, 2023-05-31:
Yes, once I'm fully finished I will publish it, got small tuning left. Solved bottom using threaded m2 inserts, so bottom is screwed on.

DREAM669, 2023-06-02:
> Yes, once I'm fully finished I will publish it, got small tuning left. Solved bottom using threaded m2 inserts, so bottom is screwed on.

Can you share the CAD file of the watch with me first? It's really beautiful! I am very curious about its structure. Thank you.

DREAM669, 2023-06-02:
Especially for your buttons, are the buttons ready-made? Or 3D printing?

jakkra, 2023-06-02:
Still printed, I did not manage to find any suitable to buy.

jakkra, 2023-06-27:
Uploaded now

closed 2023-06-27 by jakkra