ZSWatch/ZSWatch #35: Split hardware and firmware repositories

AlmazShai, 2023-08-14:
## Prerequisites
Since the ZSWatch hardware and firmware doesn't have anything in common (expect of fact that all of this related to one final product) I purpose to split it to different repositories. By growing project size and difficulties to maintain hw and fw in one repository we'll definitely end up with splitting them. I suggest doing it in early state of project before it's got too painful. By splitting the repository following advantages:
1. Possibility to setup independent pipelines
2. Follow some branching model for firmware
3. Track and handle releases/changelog clearly for firmware and hardware 
4. Configure ZSWatch firmware repo as a manifest file in proper way, which provide possibility to:
-- clone project and all dependencies via single command line,
-- checkout certain version of Zephyr and modules
-- automatically apply patches to Zephyr
-- extend west command
-- ect.
5. Define issues independently -> avoid mess of TODO
6. Open root directory in VSCode which provide:
-- possibility to operate with Git extension for VSCode
-- nRF Connect extension automatically loads application
-- set up workspace and firmware specific settings

## Side effects
1. Some manual merging required for all forks
2. Content of current repo will be changes, people will not see that they expect by clicking old link

## Proposal
1. Create two new repositories named "ZSWatchFw" and "ZSWatcHw" and copy related part to there.
2. Keep current repo as a ZSWatch overview, keep demos, documentation, images.
3. Configure ZSWatch manifest repo which will include all 3 repositories and have a following tree:
ZSWatch                           # zephyr top dir
   ├─ ZSWatch
   ├─ ZSWatchHw
   ├─ ZSWatchFw
   │    └─ west.yml             # Main manifest
   ├─ zephyr
   ├─ modules
   ├─ bootloader
   ... 

I don't like the idea of having hw sources in zephyr manifest, but it might be useful for someone.

jakkra, 2023-08-14:
Hi great suggestions! 

My idea initially was to take the https://github.com/zephyrproject-rtos/example-application and just add another folder in the root and call it `hardware`.  OR as you suggest point to a submodule with the ZSWatch HW. 

I'm sure your suggestion is much better, but I'd like to keep the SW and day to day development in this repo. Hardware I'm fine with moving. This is the repo people click to, and I want it easily to be seen what's happening, commits, etc. etc.

My proposal:
- Use example-application, copy in all code into that to match that structure. 
- I convert all stuff here https://github.com/users/jakkra/projects/1 to issues (there is a feature that does that) and add some extra description/context.
    - Convert some relevant TODSs int the code to proper issues.
- Either keep HW here, or put in a new submodule.
    - I'm not fully familiar with how west modules compare to git submodules, but I'd like it to be easily browsable to like below.
    -  [image]

What do you think?

AlmazShai, 2023-08-14:
Fully agree with that, I hope this engage more contributors. I like the idea make this repo as main firmware repository.
> This is the repo people click to, and I want it easily to be seen what's happening, commits, etc. etc.

I don't like the keeping hardware in this repo mainly because I believe that hardware also should have it's own versioning and branch model. For now it's quite difficult to find v1 revision for example, but we can improve that by using simple tagging and following simple man/develop model.

Making hardware as a submodule also have some cons:
- west could not handle submodules
- by using submodule you always point to specific commit of child repo and it' quite difficult to track last commit in repo (always need to commit of selecting new submodule commit in parent repo). In contrast, a manifest can refer to a commit, branch or tag, which is very handy.

My main intention is not to restructure the repository (that would be nice too, of course) but to make a **clear versioning**.

jakkra, 2023-08-14:
@AlmazShai also I think today the src folder have too many files in it and it makes it harder to find/navigate also it looks more daunting for new people. Maybe as part of this since files may need to move anyway, we could restructure folder the structure a bit.

AlmazShai, 2023-08-14:
> @AlmazShai also I think today the src folder have too many files in it and it makes it harder to find/navigate also it looks more daunting for new people.
Yes you are absolutely right. But it need to be done carefully, with reworking CMake files. 

> Maybe we can have a "hardware" folder and just a Readme with link or something to that repo and have it as a west module.

I see you intention to make it visible for contributor and this is right direction!!! We can create HARDWARE.md or folder, whatever you want.

jakkra, 2023-08-14:
@AlmazShai let me know if this is something you'd like to do. Otherwise I'll add it to my list and work on it later this week.

AlmazShai, 2023-08-15:
I would love to organize main manifest file!

jakkra, 2023-08-21:
I have started with some of the above discussed tasks:

- I have now converted all SW tasks to issues: https://github.com/jakkra/ZSWatch/issues
- Issues have been labeled and I tried giving all a bit more context/info in the description.
- I have prepared the new repositories for the hardware. All hardware will in the next few days be removed from this main repo and put into new ones, one for the ZSWatch main PCB and one for the dock pcb

jakkra, 2023-10-01:
Done!

closed 2023-10-01 by jakkra