ZSWatch/ZSWatch #44: Auto patches to Zephyr when compiling

jakkra, 2023-08-21:


vChavezB, 2023-08-22:
Golioth has a script to run a patch command when using the manifest file

https://github.com/golioth/golioth-zephyr-sdk/blob/main/scripts/west_commands/patch.py

and you can see it in action here

https://github.com/golioth/golioth-zephyr-sdk/blob/cc146c1e0d4f2c666a37da3b6de7cc5902367484/west-ncs.yml#L15

This issue would be related to first setting up a manifest file as mentioned in https://github.com/jakkra/ZSWatch/issues/35

ldab, 2023-11-15:
@jakkra I can only see “ app/zephyr_patches/sdl_upscale.patch” is there another one?

Kampi, 2023-11-15:
@ldab we have a patch for the nPM1300 driver which is necessary for later revisions with the new power management. It´s not checked in in Git, so you can add it if you want

[ntc_fix.patch](https://github.com/jakkra/ZSWatch/files/13367904/ntc_fix.patch)

jakkra, 2023-11-15:
@ldab do note that this sdl patch does not work anymore, that code was refactored. It's not necessary, so it can be removed.

ldab, 2023-11-15:
> @ldab do note that this sdl patch does not work anymore, that code was refactored. It's not necessary, so it can be removed.

both `kscan_sdl.c` and `display_sdl.c` are not required anymore? How to make the screen bigger?

jakkra, 2023-11-15:
> > @ldab do note that this sdl patch does not work anymore, that code was refactored. It's not necessary, so it can be removed.
> 
> both `kscan_sdl.c` and `display_sdl.c` are not required anymore? How to make the screen bigger?

Well they are needed to scale it up, but requires re-doing this patch. But current default not scaled is useable also.

Kampi, 2023-11-23:
@jakkra @ldab 
CI/CD is reporting this

[image]

jakkra, 2023-11-23:
> @jakkra @ldab CI/CD is reporting this
> 
> [image]

I got a fix incoming.

ldab, 2023-11-24:
Strange, main gives me:

```
-- west build: generating a build system
patching file modules/lvgl/input/lvgl_common_input.c
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored
patching file modules/lvgl/input/lvgl_common_input.h
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored
patching file drivers/sensor/npm1300_charger/npm1300_charger.c
Reversed (or previously applied) patch detected!  Assume -R? [n] 
Apply anyway? [n] 
Skipping patch.
1 out of 1 hunk ignored
```

In order to reproduce I tried:

1. Clean the build folder
2. Pristine build
3. Build again
4. Clean changes on Zephyr repo

Can't reproduce it

jakkra, 2023-11-24:
Fixed now and CI as good. Works locally also for me.
https://github.com/jakkra/ZSWatch/commit/694798a6fd55b2753d6025e603d2a14534c9c692

closed 2023-11-24 by jakkra