ZSWatch/ZSWatch #48: Received notifications should be propogated as zbus event. See example how to do it here: https://github.com/zephyrproject-rtos/zephyr/pull/59071/files.

jakkra, 2023-08-21:


ldab, 2023-11-15:
Is this still relevant? Otherwise it would be better to do it before #61

Kampi, 2023-11-15:
@jakkra does it make sense to propagate also the connected and disconnected callbacks via ZBUS? I can do the ZBUS change if it´s okay because the sensor subsystem modification also needed some stuff for ZBUS and then I can add the BLE stuff too.

jakkra, 2023-11-15:
Sure, but be careful of context of the notifications, I think that's why they are not yet distributed over zbus, as they can be overwritten. @Kampi read this https://discord.com/channels/720317445772017664/1042530682011926631/1113790721133129738 first

jakkra, 2023-11-15:
> Is this still relevant? Otherwise it would be better to do it before #61 

Hmm, they touch the same code, but the order doesn't really matter. If notifications are sent over zbus or callback it's just a few lines of common code, so I think they can be worked on in parallel also. Or any order.

Kampi, 2023-11-15:
IMO you can start working on the iOS stuff because the sensor testing and the ZBUS modification will need some time.

jeffwelder-ellenbytech, 2023-11-15:
If you have not checked out the new message subscriber feature of zbus it might be worth looking at. 😄 

`ZBUS_MSG_SUBSCRIBER_DEFINE()` 
https://docs.zephyrproject.org/latest/services/zbus/index.html

Kampi, 2023-11-15:
Thanks @jeffwelder-ellenbytech 

I will take a look at it.

jakkra, 2023-11-15:
> If you have not checked out the new message subscriber feature of zbus it might be worth looking at. 😄
> 
> `ZBUS_MSG_SUBSCRIBER_DEFINE()` https://docs.zephyrproject.org/latest/services/zbus/index.html

Looks interesting, but from what I see it requires a thread with stack for each subscriber, which is what I want to avoid as it wastes alot of RAM. Best would be to always use the system work queue for all messages, but not exactly sure how to implement that.

Kampi, 2023-11-20:
@ldab I added ZBUS to the `BLE_enhancements#124` branch and tested it with GadgetBridge and a subscriber and it looks good.

[image]

I´ve added it also to the Apple Notification service. If you have time, can you please test it? :)

ldab, 2023-11-20:
It looks like the content/title/body is missing or got lost, it was supposed to print

mobile phone
Incoming call

[image]

jakkra, 2023-11-20:
> @Kampi maybe this change you did that caused this issue https://github.com/jakkra/ZSWatch/blob/main/app/src/main.c#L358
> 
> body => sender (I guess it's not populated the same in iOS code)

I'm fixing it now

Kampi, 2023-11-20:
Yeah was a bad testing by me because I only tested it with Gadgetbridge

closed 2023-12-05 by Kampi