ZSWatch/ZSWatch #60: Algorithm to detect facing away watch to turn off faster intstead of 30 second timeout

jakkra, 2023-08-21:


Kampi, 2023-11-23:
@jakkra what do you think? Can we use the "Flick-in" / "Flick-out" gestures of the IMU for it?

[image]

The BMI270-Plus driver supports them. So maybe there is no algorithm needed.

jakkra, 2023-11-23:
@Kampi My experience with the various gestures is that they sometimes trigger randomly also. But good idea with the "flick out", we can try it and see how it works :+1:

Kampi, 2023-11-23:
I didn't understand the gesture completely during my tests, but I think it was because I have used the wrong axis setup. We have to take care that the coordinate system of the different watch revisions is the same as in the datasheet to test it properly.