ZSWatch/ZSWatch #97: Voice Recording

RustoMCSpit, 2023-10-29:
i know this is a longshot idea but the watch being able to record and locally store voice recordings would be great

Kampi, 2023-10-29:
Hi @RustoMCSpit,

we plan to add an I2S microphone in revision 5. With that, it´s technically possible to record mono audio and store it in the external flash memory of the watch. USB is also coming with revision 5, so you can i. e. download the file via USB or transmit it via Bluetooth. But we aren´t sure when we can add such a feature to the software.

RustoMCSpit, 2023-10-29:
> transmit it via Bluetooth

what about warpinator?

Kampi, 2023-10-29:
Hi @RustoMCSpit,

I´m not familiar with it, but the transport layer can be chosen freely when the hardware requirements are added to the electronic and every user can change or add it as he wishes :)

RustoMCSpit, 2023-10-29:
> Hi @RustoMCSpit,
> 
> I´m not familiar with it, but the transport layer can be chosen freely when the hardware requirements are added to the electronic and every user can change or add it as he wishes :)

it's like bluetooth but foss! if it's possible could support for it be built in when the time comes round

jakkra, 2025-07-03:
Closing for now. We will do new tickets when we set the scope for the microphone on the new ZSWatch reworked version.

jakkra, 2025-09-09:
Fixed in https://github.com/ZSWatch/ZSWatch/pull/439 and an upcoming PR adds some more nice mic stuff :)

QiangF, 2025-11-07:
With the microphone, speech recognition and input text with voice is a nice feature to have. The watch will be like a little secretary on hand to save any ideas easily anywhere.