ZSWatch/ZSWatch #109: Notification popup stops showing sometimes

jakkra, 2023-11-16:
It seems as the watch sometimes comes in a state where it thinks a popup notification is already shown and won't show the popup when new notifcations comes in.

From attaching the debugger when it happened it seems as this is not set to NULL for some reason: https://github.com/jakkra/ZSWatch/blob/main/app/src/ui/notification/zsw_popup_notifcation.c#L116

jakkra, 2023-12-05:
@Kampi it's fixed? Trying to fix this right now hehe

Kampi, 2023-12-05:
I couldn´t reproduce it until now. Feel free to reopen it when it happens again :)

jakkra, 2023-12-06:
> I wasn´t able to reproduce it until now. Feel free to reopen it when it happens again :)

Still there, super hard to reproduce, but I have the issue multiple times per day...

jakkra, 2024-02-29:
Think it's fixed in https://github.com/jakkra/ZSWatch/pull/229 closing for now

closed 2024-02-29 by jakkra