ZSWatch/ZSWatch #112: Common API for communication back to phone apps

jakkra, 2023-11-16:
This can remove `#ifdefs` in all application code that sends information back to a phone.

Suggested to create a header file with necessary functions such as "next track" etc.

Then we have two implemementations for it, one for Gadgetbridge and one for iOS.
We choose which one to compile depending on if the compile for Android or iOS

ldab, 2023-11-18:
I think the FW should be build for both phones and from the connection it can know what type of command to use, no ifdefs, no different headers

jakkra, 2023-11-18:
> I think the FW should be build for both phones and from the connection it can know what type of command to use, no ifdefs, no different headers

Yes true, can be dynamic, it's easily detectable as we know directly if the phone have the apple services, it's an iPhone.

ldab, 2023-11-18:
I was thinking we can use the company identifier to determine the device.

alternatively hook the play/pause etc action to a Zbus fire and forget, down the line the GATT client will not be able to write if the service is not available, sanitising it that way, very simple, I do not see the downside of doing it that way

Kampi, 2023-11-18:
Maybe we can add both variants? Build up Android and iOS separately, enable both by default, and let the user decide to disable one of those to save code space. IMHO for me iOS isn´t relevant and I don´t need it, because I´m using Android.

jakkra, 2023-11-18:
I don't care which way, the only reason I see to have it a compile thing is that it will not work with upstream Zephyr and need NCS, and that is handy as sometimes when needed to try some patch etc.
But should not make it much more complicated to keep them compile options but allow both, just need to think a bit how to structure it.

ldab, 2023-11-18:
I haven’t tried but theoretically one can set the zephyr downstream version if needed on the west.yml

jakkra, 2023-11-18:
You can, but often it won't compile as the ncs stuff is not compatilbe with newer zephyr