ZSWatch/ZSWatch #114: Unable to burn code, unable to recognize MCU

DREAM669, 2023-11-17:
Hello, I have created a PCB after referencing your first schematic diagram. Of course, this version uses TXB0104 with external Flash, but there is a phenomenon: the first time it can be successfully burned, the screen is lit, and the touch is invalid (caused by TXB0104); But there was a burning failure later, and the MCU could not be recognized. Have you ever encountered it? Thank you very much!

Kampi, 2023-11-17:
Hi @DREAM669 

can you please paste in the error message so we can take a look at it?

DREAM669, 2023-11-20:
-- west flash: using runner nrfjprog
-- runners.nrfjprog: Flashing file: e:\ZSWatch-main_newbard\app\build\zephyr\merged_domains.hex
-- runners.nrfjprog: e:\ZSWatch-main_newbard\app\build\zephyr\merged_domains.hex targets both nRF53 coprocessors; splitting it into: e:\ZSWatch-main_newbard\app\build\zephyr\GENERATED_CP_NETWORK_merged_domains.hex and e:\ZSWatch-main_newbard\app\build\zephyr\GENERATED_CP_APPLICATION_merged_domains.hex
[error] [ Client] - Encountered error -90: Command read_memory_descriptors executed for 15 milliseconds with result -90
[error] [ Worker] - Can't read memory descriptors, ap-protection is enabled.
[error] [ Client] - Encountered error -90: Command erase_file executed for 92 milliseconds with result -90
[ ######               ]   0.000s | Erase file - Check image                                                       [ #####                ]   0.000s | Check image validity - Initialize device info                                  [ ##########           ]   0.000s | Check image validity - Check region 0 settings                                 [ ###############      ]   0.000s | Check image validity - block 1 of 2                                            [error] [  nRF53] - Failed while detecting device memory block protection status!
[error] [  nRF53] - Failed while erasing device. -90: Access protection is enabled, can't read block protection state.
[error] [ Worker] - Access protection is enabled, can't read block protection state.
ERROR: The operation attempted is unavailable due to readback protection in
ERROR: your device. Please use --recover to unlock the device.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
ERROR: runners.nrfjprog: Flashing failed because the target must be recovered.
  To fix, run "west flash --recover" instead.
  Note: your target is an nRF53; all flash memory for both the network and application cores will be erased prior to reflashing.
FATAL ERROR: command exited with status 24: nrfjprog --program 'e:\ZSWatch-main_newbard\app\build\zephyr\GENERATED_CP_NETWORK_merged_domains.hex' --sectorerase --verify -f NRF53 --coprocessor CP_NETWORK --snr 1050050002

 *  The terminal process terminated with exit code: 24. 
 *  Terminal will be reused by tasks, press any key to close it.

DREAM669, 2023-11-20:
Hi,@Kampi 
I'm sorry, I was too busy the other day and forgot about it. Could you please help me check this error? Thank you very much!

jakkra, 2023-11-20:
@DREAM669 You have the error there in your log:
```
ERROR: The operation attempted is unavailable due to readback protection in
ERROR: your device. Please use --recover to unlock the device.
NOTE: For additional output, try running again with logging enabled (--log).
NOTE: Any generated log error messages will be displayed.
ERROR: runners.nrfjprog: Flashing failed because the target must be recovered.
To fix, run "west flash --recover" instead.
```

Did this not work?
`west flash --recover`

If it doesn't work try running those manually instead of through west:
```
nrfjprog -f NRF53 --recover --coprocessor CP_NETWORK
nrfjprog -f NRF53 --recover
```

DREAM669, 2023-11-21:
Thank you very much. The problem has been resolved.

closed 2023-11-21 by DREAM669