ZSWatch/ZSWatch #124: Rework pairing

Kampi, 2023-11-19:
**ToDo**

- [ ] Add timeout for enabled pairing
- [ ] Make it visible in the GUI that pairing is enabled / disabled

Kampi, 2023-11-21:
@ldab 

I split the Gadgetbridge related code up into a separate source file.

https://github.com/jakkra/ZSWatch/tree/BLE_enhancements%23124

ldab, 2023-11-21:
Will check the PR tomorrow 👊

Kampi, 2023-11-25:
We are changing the communication flow for notifications to this
[image]

This allows us to create and push internal notifications to the notification manager. The notification manager is the main instance in handling all notifications.

@jakkra @ldab 
I pushed the latest version of the notification rework. Feel free to test this version and give some feedback :)
Not 100% nice now, but it´s working.

**Changes so far**

- New layout
- Different icons for WhatsApp, Mail, etc.
- Title and body field
- Scrollable body (not sure if I will keep it)
- Age of the notification
- Scrolling improvement
- Notifications will be added to the notification app when the app is open

**Known changes**

- Convert the age into a nice time format
- Scrolling of the body field is strange
- Sometimes the body field will be focused
- Scrolling of the notification list can be a little bit weird. I guess because of the scrolling body field. Need to check it in detail
- Hide notification popup when the app is open. Here we should think about an API to get the app status

[image]

ldab, 2023-11-25:
The latest commit does not seem to build on CI, I haven’t looked as why, but is this ready to feedback?

Kampi, 2023-11-25:
@ldab the build fails because of the style.

jakkra, 2023-11-25:
@Kampi Running it today!
Some quick things:
- Send some notfications
  - Enter notification list
  - Exist -> crash

- Longer notifications the body can't be seen

Kampi, 2023-11-25:
>   * Enter notification list
>   * Exist -> crash
> * Easy to accidentally remove the notification when scrolling it

@jakkra fixed :)

closed 2023-12-05 by jakkra