ZSWatch/ZSWatch #130: Upgrade to use astyle 3.4 in ci

jakkra, 2023-11-20:
The current astyle version in apt-get is 3.1 but the one users typically download from the web is the latest 3.4.
There is some difference in 3.1 vs 3.4 in the way they format code, hence causing ci issues when user and ci don't have matching astyle version.

Also add instructions for Linux users how to download and install astyle 3.4 since apt-get uses an old version.

jakkra, 2023-11-28:
@ldab We do astyle install through apt-get https://github.com/jakkra/ZSWatch/blob/main/.github/workflows/build.yml#L27.

This is unrelated to Zephyr version etc.

ldab, 2023-11-28:
aaaaaaaaaaaaaaaaaaaaaa

Kampi, 2023-12-27:
@jakkra I added the build and installation of AStyle 3.4 to the development container. Can you add these commands to the CI/CD pipeline too?

```
wget https://sourceforge.net/projects/astyle/files/astyle/astyle%203.4/astyle-3.4.10.tar.bz2/download -O astyle.tar.bz2
tar -xf astyle.tar.bz2
cd astyle-3.4.10
mkdir as-gcc-exe
cd as-gcc-exe
cmake ../
make
```

closed 2023-12-29 by jakkra