ZSWatch/ZSWatch #138: Integrate Bosch BSX Lite library for 6-DOF sensor fusion

jakkra, 2023-11-22:
With this we can utilize Bosch library for fusion for Accel + Gyro to get Roll, Yaw  and Pitch angles.

This is what the library can output for us, can be used for some nice stuff!
[image]

https://www.bosch-sensortec.com/software-tools/software/sensor-fusion-software-bsx/

ldab, 2023-11-22:
On it 🫡

ldab, 2023-11-22:
How do we deal with licensing? Bosch ask for license agreement, perhaps we can't share openly? like people need to go there and click with their info?

jakkra, 2023-11-22:
Yes off by default (kconfig) and user need to download and put in a specific folder. For the BSEC2 it's solved a bit easier, even though I think it's the same license. But bosch have it on Github, which means we cannot automatically clone it, but using a west command see here https://github.com/jakkra/ZSWatch/blob/main/GETTING_STARTED.md#use-bosch-bsec2 and this is how nordic also does it, so assumed to be fine.

For now at least start with user need to download and put it in a specific folder, and then enable using kconfig

ldab, 2023-11-22:
> Yes off by default (kconfig) and user need to download and put in a specific folder. For the BSEC2 it's solved a bit easier, even though I think it's the same license. But bosch have it on Github, [...]

Cool, URL on readme is broken btw

jakkra, 2023-11-22:
> Cool, URL on readme is broken btw

Yep, already fixed :)

ldab, 2023-11-23:
BSXlite is not on GitHub. Do you have some features that needs it now? If not then I’ll work on the speedometer thing before as 
1. this BSXlite may need more thoughts 
2. and I also think it will clash with the pedometer
3. It does not give the heading speed anyway

jakkra, 2023-11-23:
> BSXlite is not on GitHub. Do you have some features that needs it now? If not then I’ll work on the speedometer thing before as
> 
> 1. this BSXlite may need more thoughts
> 2. and I also think it will clash with the pedometer
> 3. It does not give the heading speed anyway

Yeah no problem, nothing really that needs it for now.

jakkra, 2023-11-24:
Just found this: https://github.com/xioTechnologies/Fusion

It's a sensor fusion library that also takes magnetometer data (compared with that BSX Lite does). 

That would give us absolute orientation, not just relative which BSX Lite does. Also this library is open source. Utilizing a full 9-Axis.

Looks quite undertandable and seems quite easy to use: https://github.com/xioTechnologies/Fusion/blob/main/Examples/Advanced/main.c 

More info: https://x-io.co.uk/open-source-imu-and-ahrs-algorithms/

ldab, 2023-11-24:
Sounds promising, generally those sensor fusion needs raw sensor input quite frequently, do you think it would interfere with Pedometer?

jakkra, 2023-11-24:
> Sounds promising, generally those sensor fusion needs raw sensor input quite frequently, do you think it would interfere with Pedometer?

Why would is clash with pedometer, that's just a built in feature of the IMU. We fetch accel and gyro data simultaniously.

ldab, 2023-11-24:
> > Sounds promising, generally those sensor fusion needs raw sensor input quite frequently, do you think it would interfere with Pedometer?
> 
> Why would is clash with pedometer, that's just a built in feature of the IMU. We fetch accel and gyro data simultaniously.

I don't know just guessing, that if we change the rate to say 100Hz, the pedometer function does not work or something like that

jakkra, 2023-11-24:
> I don't know just guessing, that if we change the rate to say 100Hz, the pedometer function does not work or something like that

I think it's fine, I only saw requirements on minimal rate.

jakkra, 2024-02-28:
Closing as it's implemented using Fusion library.

closed 2024-02-28 by jakkra