ZSWatch/ZSWatch #140: Jlink&IIC

DREAM669, 2023-11-23:
Hello, I have encountered two issues.
First question: How can I use JLink for debugging in VS code? What are the minimum pins that JLink needs to connect to?
The second question: Reading sensor information through IIC and calling the IIC read interface resulted in an incorrect return value. What is the approximate reason?

Kampi, 2023-11-23:
Hi @DREAM669 
1) SWD needs
- SWDCLK
- SWDIO
- GND
- VTarget
- VCC on the target
You can debug by using the Debug function of the nRF Connect Plugin in VS Code (NOT the regular debug!)

2) Hard to say. Can you please share your code?

DREAM669, 2023-11-23:
[image]

DREAM669, 2023-11-23:
I clicked the debug button, then it showed no connected device found.

Kampi, 2023-11-23:
It looks like that no SWD target is found. Also your debugger looks strange because of the loading circle. 
Please do the following:
- What kind of debugger do you use?
- Does the debugger gets loaded correctly when no target is attached? Restart VS Code to be sure
- Can you expand the debugger menu and see a target?

DREAM669, 2023-11-23:
> It looks like no SWD target is found. Also, your debugger looks strange because of the loading circle. Please do the following:
> 
> * What kind of debugger do you use?
I am using J-Link as shown in the following image.
[image]

> * Does the debugger get loaded correctly when no target is attached? Restart VS Code to be sure
When I only connect J-Link to the computer without connecting it to the target board, it looks like this.
[image]

> * Can you expand the debugger menu and see a target?
When I connect four pins（SWDCLK，SWDIO，GND，VTarget） to J-Link, it looks like this.
[image]

> * Does the power supply of the target work?
> Yes, the target board is powered properly as the screen displays normally.

Kampi, 2023-11-23:
Okay that looks good so far. Can you program the MCU?

DREAM669, 2023-11-23:
when I click Flash Button, it shows "No connected device found with serial number 1050050002.
Either unlink the device from the build or connect that device."
I Jlink serial number is 5930563, do I need change this number?

jakkra, 2023-11-23:
> when I click Flash Button, it shows "No connected device found with serial number 1050050002. Either unlink the device from the build or connect that device." I Jlink serial number is 5930563, do I need change this number?

No, serial number is fixed, can't change.

Please use JFlash Lite and try do an erase in it. Also try flashing the built hex files directl using JFlashLite to see if it works. This will tell us if the debugger is working fine.

DREAM669, 2023-11-23:
I previously used the nRF5340DK board for downloading programs to the ZSwatch, and it worked fine. The serial number (SN) of the nRF5340DK is 1050050002.

DREAM669, 2023-11-23:
> No, serial number is fixed, can't change.
>
> Please use JFlash Lite and try do an erase in it. Also try flashing the built hex files directl using JFlashLite to see if it works. This will tell us if [...]

I am able to download programs successfully using the command-line tool, which indicates that my J-Link is functional. However, I am unable to download programs and debug within VS Code.

jakkra, 2023-11-23:
Ok, please try this:
[image]

Try both of those, see if one of them works
[image]

DREAM669, 2023-11-23:
@jakkra
PS E:\ZSWatch-main_newbard\app> nrfjprog -f NRF53 --recover
Recovering device. This operation might take 30s.
Erasing user code and UICR flash areas.
Writing image to disable ap protect.

jakkra, 2023-11-23:
> @jakkra
> PS E:\ZSWatch-main_newbard\app> nrfjprog -f NRF53 --recover
> Recovering device. This operation might take 30s.
> Erasing user code and UICR flash areas.
> Writing image to disable ap protect.

That looks correct, is there a question?

DREAM669, 2023-11-23:
> > @jakkra
> > PS E:\ZSWatch-main_newbard\app> nrfjprog -f NRF53 --recover
> > Recovering device. This operation might take 30s.
> > Erasing user code and UICR flash areas.
> > Writing image to disable ap protect.
> 
> That looks correct, is there a question?

My problem is that I want to use Jlink to debug the program in VS code, but I get an error about "No connected device found with serial number 1050050002. Either unlink the device from the build or connect that device."

jakkra, 2023-11-23:
> My problem is that I want to use Jlink to debug the program in VS code, but I get an error about "No connected device found with serial number 1050050002. Either unlink the device from the build or [...]

Did you try changing device provider as suggested?

DREAM669, 2023-11-23:
"No connected device found with serial number 1050050002. Either unlink the device from the build or connect that device."
Does this sentence mean that the debugging tool I set before is nRF5340DK, and I need to set the debugging tool to Jlink?

DREAM669, 2023-11-23:
My nRF5340DK cannot work, so I can only use Jlink.

Kampi, 2023-11-23:
> My problem is that I want to use Jlink to debug the program in VS code, but I get an error about "No connected device found with serial number 1050050002. Either unlink the device from the build or [...]

1050050002 is your DK? So the used device in VS Code is your DK and not the JLink. Can you please unplug the DK and plug in only the JLink to make sure that there is no issue?

DREAM669, 2023-11-23:
Yes, 1050050002 is nRF4340DK.  I have unpluged the DK.
I guess what we have to do now is how to implement "unlink the device from the build"

jakkra, 2023-11-23:
> Yes, 1050050002 is nRF4340DK. I have unpluged the DK. I guess what we have to do now is how to implement "unlink the device from the build"

Just click here to link/unlink
[image]

DREAM669, 2023-11-24:
> > Yes, 1050050002 is nRF4340DK. I have unpluged the DK. I guess what we have to do now is how to implement "unlink the device from the build"
> 
> Just click here to link/unlink [image]

OK! 
This problem is solved. Thank you so much！@jakkra@Kampi

Kampi, 2023-11-24:
Glad to hear that! Enjoy the hacking 😀

closed 2023-11-24 by Kampi