ZSWatch/ZSWatch #145: Question: why base on Zephyr rather than NCS

pellepl, 2023-11-24:
This is not a bug, but a question:
what was the reason for choosing to base the project on Zephyr rather than on Nordics Connect SDK version of Zephyr? I read it was due to some code not yet merged into NCS, but I'd like to know more specific.

And multitudes of gratitude for open sourcing this great project! Impressive!

jakkra, 2023-11-24:
Hi, the idea was that pure Zephyr is more open than NCS. NCS contains Nordic licensed code which may only run on Nordic HW. 
Which means we could port to additional HW if we want. 

But anyway, actually right now we are using NCS 2.5 default, as for iOS there is a bunch of stuff not part of stock Zephyr, so it's just more handy to stay with NCS and not fight it. NCS 2.5 contains the fixes needed and that it's now default. Could be some documentation is not updated to say this, but as you can see here: https://github.com/jakkra/ZSWatch/blob/main/app/west.yml NCS is default.

pellepl, 2023-11-24:
Cheers!

closed 2023-11-24 by pellepl