ZSWatch/ZSWatch #153: Common location for images and icon

Kampi, 2023-11-26:
- Should be placed in the external flash
- Should prevent fragmentation and double images to save space
- QSPI preferred (how do we want to handle the "normal" SPI?)

jakkra, 2023-11-27:
Need to experiment a bit with v4 and QSPI, if we can put everything in external it would be great.
For v3 I'm not sure exactly how to handle it as stuff on watchface must be in internal flash, otherwise it's too slow.

Will think abit about what's the best approach. 

Also another issue is that some of the .c images files I don't know the source of, and they need to be converted to .bin to work in external flash. So will need some thinking.

Kampi, 2023-11-27:
I think it's enough to think about how to implement it in rev 4 with all the infrastructure. A Backport to rev 3 can be done later if there is a need to do it. For now it looks okay 👍

closed 2024-02-27 by jakkra