ZSWatch/ZSWatch #164: API to let apps disable on / off of the screen

Kampi, 2023-11-28:


Kampi, 2023-11-28:
@jakkra I´m not sure about the priority. Do you have an idea?

jakkra, 2023-11-28:
@Kampi Low, but whenever someone needs it, we can implement it. Should be a simple one. Also possible right now for an app to workaround it just call the [zsw_power_manager_reset_idle_timout](https://github.com/jakkra/ZSWatch/blob/main/app/src/managers/zsw_power_manager.c#L115 ) every 10 seconds or something. So this is actually achievable right now

Kampi, 2023-11-28:
> @Kampi Low, but whenever someone needs it, we can implement it. Should be a simple one. Also possible right now for an app to workaround it just call the [...]

@ldab maybe a good solution for you?

ldab, 2023-11-28:
> @ldab maybe a good solution for you?

I find it ugly, but it is a solution. I think I will wait for the API as the app itself needs some tunning