ZSWatch/ZSWatch #171: NTS0104GU12 questions

DREAM669, 2023-11-30:
Currently, I am facing a problem. You did not connect a pull-up resistor to your NTS0104, but I found that the signal of IIC without a pull-up resistor is unstable. After connecting the pull-up resistor, the voltage level is relatively normal.Thank you for your reply!

Kampi, 2023-11-30:
Hi @DREAM669 

which pull-up resistors do you mean? The IC has internal pull-up resistors attached to its inputs and outputs. Please check the [datasheet](https://www.nxp.com/docs/en/data-sheet/NTS0104.pdf) on page 15 topic 14.2.

DREAM669, 2023-12-01:
Okay, let me sort out the current issues.
1.After connecting the connector to the display screen, measure the IIC voltage. SDA and SCL changed from normal 3.3V to 2.7V and 2.5V
2. After downloading the program, the Touch function does not work. Is it due to voltage issues that the Touch chip cannot be driven?
Thank you very much for your reply!

jakkra, 2023-12-01:
> Okay, let me sort out the current issues. 1.After connecting the connector to the display screen, measure the IIC voltage. SDA and SCL changed from normal 3.3V to 2.7V and 2.5V 2. After downloading [...]

What PCB version did you order? Because we never had a BOM with this wrong logic shifter.

DREAM669, 2023-12-01:
Initial version PCB. The version using TXB0104 frequency conversion chip. But I soldered NTS0104 onto it again, replacing TXB0104. May I ask, after you connect the display screen, is the voltage of IIC still 3.3V? No decline? thank you.

jakkra, 2023-12-01:
> Initial version PCB. The version using TXB0104 frequency conversion chip. But I soldered NTS0104 onto it again, replacing TXB0104. May I ask, after you connect the display screen, is the voltage of IIC still 3.3V? No decline? thank you.

Please provide the commit you used

DREAM669, 2023-12-01:
I am using your first version. You didn't encounter this situation at the time, did you?

DREAM669, 2023-12-01:
Okay, I understand what to do now. I will try to compare. There is only one connector on the PCB. Thank you.

jakkra, 2023-12-01:
@DREAM669 I made many different version, so I'm not sure which one 

> Okay, I understand what to do now. I will try to compare. There is only one connector on the PCB. Thank you.

Ok, then you are here. Just see if I changed something else on the PCB, if not you may have a soldering issue.
[image]

jakkra, 2023-12-04:
> May I ask if the voltage converted by NTS0104 after connecting the display screen is 3.3V? Or is there a decrease? Thank you for your reply!

Will measure when I have a chance. May be a day or two.

DREAM669, 2023-12-13:
Finally, it was discovered that the pin pin definition was incorrect. Thank you

closed 2023-12-13 by DREAM669