ZSWatch/ZSWatch #202: error about compiling the latest code in vscode

DREAM669, 2023-12-12:
I downloaded the latest code, but there was an error when compiling in vscode. Can you help me find out the reason?

Executing task: nRF Connect: Build: app/build (active) 

Building app
C:\Windows\system32\cmd.exe /d /s /c "west build --build-dir e:/git/ZSWatch/app/build e:/git/ZSWatch/app"

-- west build: generating a build system
Loading Zephyr default modules (Zephyr base (cached)).
-- Application: E:/git/ZSWatch/app
-- CMake version: 3.20.5
-- Cache files will be written to: C:/ncs/v2.4.0/zephyr/.cache
-- Zephyr version: 3.3.99 (C:/ncs/v2.4.0/zephyr)
-- Found west (found suitable version "1.0.0", minimum required is "0.7.1")
-- Board: zswatch_nrf5340_cpuapp, Revision: 3
-- Found host-tools: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.0 (C:/ncs/toolchains/31f4403e35/opt/zephyr-sdk)
-- Found BOARD.dts: E:/git/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp.dts
-- Found devicetree overlay: E:/git/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp_3.overlay
devicetree error: E:/git/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp_3.overlay:41 (column 24): parse error: expected number or parenthesized expression
-- In: E:/git/ZSWatch/app/build/zephyr, command: C:/ncs/toolchains/31f4403e35/opt/bin/python.exe;C:/ncs/v2.4.0/zephyr/scripts/dts/gen_defines.py;--dts;E:/git/ZSWatch/app/build/zephyr/zephyr.dts.pre;--dtc-flags;'-Wno-unique_unit_address_if_enabled';--bindings-dirs;C:/ncs/v2.4.0/nrf/dts/bindings;E:/git/ZSWatch/app/dts/bindings;C:/ncs/v2.4.0/zephyr/dts/bindings;--header-out;E:/git/ZSWatch/app/build/zephyr/include/generated/devicetree_generated.h.new;--dts-out;E:/git/ZSWatch/app/build/zephyr/zephyr.dts.new;--edt-pickle-out;E:/git/ZSWatch/app/build/zephyr/edt.pickle;--vendor-prefixes;C:/ncs/v2.4.0/nrf/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;E:/git/ZSWatch/app/dts/bindings/vendor-prefixes.txt;--vendor-prefixes;C:/ncs/v2.4.0/zephyr/dts/bindings/vendor-prefixes.txt
CMake Error at C:/ncs/v2.4.0/zephyr/cmake/modules/dts.cmake:276 (message):
  gen_defines.py failed with return code: 1
Call Stack (most recent call first):
  C:/ncs/v2.4.0/zephyr/cmake/modules/zephyr_default.cmake:115 (include)
  C:/ncs/v2.4.0/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/ncs/v2.4.0/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:97 (include_boilerplate)
  CMakeLists.txt:30 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\31f4403e35\opt\bin\cmake.EXE' '-DWEST_PYTHON=C:\ncs\toolchains\31f4403e35\opt\bin\python.exe' '-Be:\git\ZSWatch\app\build' -GNinja '-Se:\git\ZSWatch\app'

 *  The terminal process terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it.

jakkra, 2023-12-12:
This is the compilation of the same scenario in CI:
https://github.com/jakkra/ZSWatch/actions/runs/7171094921/job/19525297160

1. Run west update
2. Do pristine build instead
3. You have C:/ncs/v2.4.0/zephyr stuff, this is wrong, we use west workspace (zephyr is inside ZSWatch folder) and we use ncs 2.5

It's doing:
`Run west build app --build-dir zswatch_nrf5340_cpuapp@3_debug -p -b zswatch_nrf5340_cpuapp@3 -- -DOVERLAY_CONFIG=boards/debug.conf"`

Can you try running that from the ZSWatch repo root directory

jakkra, 2023-12-12:
If you still have issues, please re-clone the repo and follow the instructions here: https://github.com/jakkra/ZSWatch/blob/main/GETTING_STARTED.md#cloning-source-code

DREAM669, 2023-12-13:
Thank you for your answer！The compilation problem is solved.
 I saw that the latest version adds an external Flash. What is this Flash used for?

jakkra, 2023-12-13:
> @jakkra Thank you for your answer！The compilation problem is solved. I saw that the latest version adds an external Flash. What is this Flash used for?

Mostly for storing image resoucres. The internal flash is 100% full, so external is needed. For example a full screen image is like 20% of internal flash. External flash also allows OTA.

closed 2023-12-13 by jakkra