ZSWatch/ZSWatch #203: Setup dev container for ZSWatch development environment

Kampi, 2023-12-15:
**Requirements**

- [ ] Neccessary extensions for VS Code
- [ ] Neccessary settings for VS Code
- [ ] Neccessary tools
  - [ ] AStyle
  - [ ] CodeChecker
- [ ] Posix build possible?

jakkra, 2023-12-15:
AStyle 3.1 I got from apt-get

Rayvert64, 2024-07-19:
It seems like this is started, but I can't seem to find a branch/pull request for this. I can help set it up, I was going to so this anyways for my own setup of this project. 🙂

I'll just start on my end and do a pull request if need be 🙂.

jakkra, 2024-07-19:
> It seems like this is started, but I can't seem to find a branch/pull request for this. I can help set it up, I was going to do this anyways for my own setup of this project. 🙂
> 
> I'll just start on my end and do a pull request if need be 🙂.

Hi, this is the PR we abandoned https://github.com/jakkra/ZSWatch/pull/211

There where too many tradeoffs etc. making the container harder to use than just installing with the nRF tools (which IMO is easy).
What I remember, debugging requires starting a GDB server and having extra handling for it in VScode etc.

I'd say don't spend too much time on it, as it turned out quite messy for us to get it running well without tradeoffs.

Kampi, 2024-07-20:
As @jakkra mentioned. But you can check out the branch on my GitHub if you want.

Rayvert64, 2024-07-21:
Cool thanks  @Kampi, but for now I'll just set it up on my side. :) 

I'll try to rework it myself for now without looking at your solution unless I get desperate hahaha.  If I have a good potential solution for most people, then I **might** do a pull request.

If I do one though, my guess is that it will probably be something more like template files?(We used to do this at my old workplace) Perhaps that could be a _decent_ solution, as it puts a bit less of a burden of responsibility for you guys. Would you guys be open to that?

jakkra, 2024-07-22:
@Rayvert64 I guess so, I don't have any specific idea how to do it.
IMO it's not really needed as I think using nRF toolchain is easy and reliable. 
But if you can come up with something nice sure go ahead.

Kampi, 2024-07-22:
@Rayvert64 what do you mean with template?

Rayvert64, 2024-07-22:
I mean having files that the user has to copy/edit to set up their files. For example, a `.env.template` file and a `devcontainer.json.template` file. These two files you would track, but you wouldn't track the actual `.env` and `devcontainer.json`files. So the template files are a bit more of a "here's a starting point, most people who edit these should get a devcontainer to work."

I know it kinda defeats the "point" of a devcontainer in terms of you can "just start to code," but I don't think I've ever seen a devcontainer not need at least _a bit_ of configuration to get working. 🫠

I don't use them for that purpose, though. I simply use them to keep my actual system "lean." 😅