ZSWatch/ZSWatch #206: sdl native posix upscaling

elmo2k3, 2023-12-21:
Hi,

in your talk you mentioned that you modified display_sdl.c to upscale the sdl output. I tried to place SDL_RenderSetLogicalSize(*renderer,600,600) to different places also in display_sdl_bottom.c but without success. Could you give me a hint?

Cheers
Bjoern

jakkra, 2023-12-21:
I removed my patch a while ago as stuff in Zephyr was moved around so it wouldn't apply. But in latest Zephyr on main it's actually implemented now. But ncs is not using that version yet. So you can probably find the commit upstream and cherry pick it.

elmo2k3, 2023-12-21:
Found it, thanks!! Nice and simple display_zoom_pct option.

jakkra, 2023-12-21:
Nice! If it applies to ZSWatch Zephyr version, feel free to add it here and do a PR https://github.com/jakkra/ZSWatch/tree/main/app/patches

jakkra, 2024-04-15:
Merged and fixed

closed 2024-04-15 by jakkra