ZSWatch/ZSWatch #222: Unable to recognize MCU and download program

DREAM669, 2024-01-30:
Hi，
@jakkra @Kampi 
Our board voltage is incorrect, and we suspect that the MCU is burnt out. Then, we replaced the MCU in the NORA-B10 module with a new one and used it in VS code to download the program. We found that the MCU could not be recognized and the code could not be burned. We executed nrfjprog - f NRF53-- recover and received the following error message.

[image]

DREAM669, 2024-01-30:
We also encountered another issue, when the battery voltage is below 3.8V, pressing the button will reboot the system. Please help take a look.

Kampi, 2024-01-30:
> We also encountered another issue, when the battery voltage is below 3.8V, pressing the button will reboot the system. Please help take a look.

The watch is working until ~3.6 V. I guess you hit the lower voltage limit and this leads to a PoR. Please charge the battery and try it again.

Kampi, 2024-01-30:
> Hi， @jakkra @Kampi Our board voltage is incorrect, and we suspect that the MCU is broken. Then, we replaced the MCU in the NORA-B10 module with a new one and used it in VS code to download the program. We found that the MCU could not be recognized and the code could not be burned. We executed nrfjprog - f NRF53-- recover and received the following error message.
> 
> <img alt="err" width="612" src="https://private-user-images.githubusercontent.com/56341360/300668809-31e091d8-c7eb-456a-801c-28d223511f6b.PNG?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1OTkxMzQsIm5iZiI6MTcwNjU5ODgzNCwicGF0aCI6Ii81NjM0MTM2MC8zMDA2Njg4MDktMzFlMDkxZDgtYzdlYi00NTZhLTgwMWMtMjhkMjIzNTExZjZiLlBORz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMwVDA3MTM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTUwNDhiNjQyYjI0YzczMzE1ZWQzNGQzZDBjNmVlMTUwMjU0OGIxNjVmNWQ4MDMwMmIxNzc1ZDIzMjUwYmI2ZWYmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.P7M_paGImIEtPjZMVnYeWahZMTv38Bg2LDPlRA41HWg">
> ![image](https://private-user-images.githubusercontent.com/56341360/300668979-85c0116d-98d2-4c9f-a6dc-fdc787aee7bb.png?jwt=eyJhbGciOiJIUzI1NiIsInR5cCI6IkpXVCJ9.eyJpc3MiOiJnaXRodWIuY29tIiwiYXVkIjoicmF3LmdpdGh1YnVzZXJjb250ZW50LmNvbSIsImtleSI6ImtleTUiLCJleHAiOjE3MDY1OTkxMzQsIm5iZiI6MTcwNjU5ODgzNCwicGF0aCI6Ii81NjM0MTM2MC8zMDA2Njg5NzktODVjMDExNmQtOThkMi00YzlmLWE2ZGMtZmRjNzg3YWVlN2JiLnBuZz9YLUFtei1BbGdvcml0aG09QVdTNC1ITUFDLVNIQTI1NiZYLUFtei1DcmVkZW50aWFsPUFLSUFWQ09EWUxTQTUzUFFLNFpBJTJGMjAyNDAxMzAlMkZ1cy1lYXN0LTElMkZzMyUyRmF3czRfcmVxdWVzdCZYLUFtei1EYXRlPTIwMjQwMTMwVDA3MTM1NFomWC1BbXotRXhwaXJlcz0zMDAmWC1BbXotU2lnbmF0dXJlPTQ3ZjY4MjdiYTg0NzdlMzRmODk1ZjI3OGZiOGQ5YmVhYzQ5OTMxN2QyZTAwMzY0MDg3MWJmNmE2Y2IxZWVhYjAmWC1BbXotU2lnbmVkSGVhZGVycz1ob3N0JmFjdG9yX2lkPTAma2V5X2lkPTAmcmVwb19pZD0wIn0.08fChfkIU6a5IhSqVtPH0rb-wHv7k3e4wWC9r5MmVyE)

Are you sure that the MCU is soldered correctly?

jakkra, 2024-01-30:
> We also encountered another issue, when the battery voltage is below 3.8V, pressing the button will reboot the system. Please help take a look.

This is expected if you are running old FW, it's the vibration motor turning on and causing the battery voltage to drop and battery supervisor will cut power.

I fixed this like a month ago to not vibrate when low battery voltage.

DREAM669, 2024-01-30:
There is another issue. I saw in your schematic that the program was downloaded with a Vbus pin, but I need to connect the+1V8 pin to the VCC pin of Jlink to download it instead of VBUS. @jakkra @Kampi

Kampi, 2024-01-30:
> There is another issue. I saw in your schematic that the program was downloaded with a Vbus pin, but I need to connect the+1V8 pin to the VCC pin of Jlink to download it instead of VBUS. @jakkra @Kampi

What do you mean? Sure you have to connect the IO voltage of the programming interface to the J-Link for it´s internal level converter. The dock connector is using a 5 V voltage as system voltage, but SWD is running on 3.3 V. So you have to connect 3.3 V to VTarget at your J-Link.

jakkra, 2024-01-30:
> What do you mean? Sure you have to connect the IO voltage of the programming interface to the J-Link for it´s internal level converter. The dock connector is using a 5 V voltage as system voltage, [...]

Hmm, I think this may be the issue. Vref should be 1v8 here not 3v3 [image]

DREAM669, 2024-01-30:
> > Hi， @jakkra @Kampi Our board voltage is incorrect, and we suspect that the MCU is broken. Then, we replaced the MCU in the NORA-B10 module with a new one and used it in VS code to download the program. We found that the MCU could not be recognized and the code could not be burned. We executed nrfjprog - f NRF53-- recover and received the following error message.
> 
> Are you sure that the MCU is soldered correctly?

I am not sure. Is there a quick and simple way to determine whether an MCU is soldered correctly?

jakkra, 2024-01-30:
> I am not sure. Is there a quick and simple way to determine whether an MCU is soldered correctly?

You can talk with it using the debugger :)

Kampi, 2024-01-30:
@DREAM669 you use the dock with an external J-Link? The dock has two use cases:

- Using internal J-Link: `Debug connector` is used for programming the STM32 -> 3.3 V is correct
- Using external J-Link: `Debug connector` is connected with `Interface`. Then you need 1.8 V. Here we have an issue for sure.

I will handle this issue in the next dock release. I´m sorry for this issue. We haven´t thought about it 👎

DREAM669, 2024-01-30:
> > > > Hi， @jakkra @Kampi Our board voltage is incorrect, and we suspect that the MCU is broken. Then, we replaced the MCU in the NORA-B10 module with a new one and used it in VS code to download the program. We found that the MCU could not be recognized and the code could not be burned. We executed nrfjprog - f NRF53-- recover and received the following error message.
> > > 
> > > 
> > > Are you sure that the MCU is soldered correctly?
> > 
> > 
> > I am not sure. Is there a quick and simple way to determine whether an MCU is soldered correctly?
> 
> You can talk with it using the debugger :) Did your wrong voltage leves get fixed by replacing MCU?

Yes，the wrong voltage leves get fixed.

DREAM669, 2024-01-30:
> @DREAM669 you use the dock with an external J-Link? The dock has two use cases:
> 
> * Using internal J-Link: `Debug connector` is used for programming the STM32 -> 3.3 V is correct
> * Using external J-Link: `Debug connector` is connected with `Interface`. Then you need 1.8 V. Here we have an issue for sure.
> 
> I will handle this issue in the next dock release. I´m sorry for this issue. We haven´t thought about it 👎
> 
> Issue: [jakkra/ZSWatch-Dock#11](https://github.com/jakkra/ZSWatch-Dock/issues/11)

 I did not use a dock. I just use a external J-Link.

Kampi, 2024-01-30:
> I did not use a dock. I just use a external J-Link.

Ah! Okay so you don´t run into the issue. Glad to hear it 👍

DREAM669, 2024-01-31:
@jakkra @Kampi 
Can the MCU in the NORA-B10 module be directly replaced with the new nRF5340? We welded and replaced the MCU in the NORA-B10 module on two boards, but we couldn't detect the MCU using Jlink. Executing nrfjprog - f NRF53-- recover -- log would result in an error.

Kampi, 2024-01-31:
Hi @DREAM669 

yes you can attach any nRF5340 dev kit or similar to the pins of the footprint of the NORA module.

jakkra, 2024-01-31:
> @jakkra @Kampi Can the MCU in the NORA-B10 module be directly replaced with the new nRF5340? We welded and replaced the MCU in the NORA-B10 module on two boards, but we couldn't detect the MCU using Jlink. Executing nrfjprog - f NRF53-- recover -- log would result in an error.

But it would be safer to just replace the whole module, or easier at least.

closed 2024-02-27 by DREAM669