ZSWatch/ZSWatch #233: Support for NFC tag emulation

nullasaservice, 2024-03-14:
First of all, nice work you have done so far!

One thing I'm looking for right now is a smartwatch that could emulate NFC tags of any kind, due to commercial smartwatches (overall) not being able to do so.

The main NFC tag type that comes to mind is MIFARE Classic 1k, which is commonly used as an access card to lots of places. It would be cool to have a chip that could clone and emulate those types of cards.

Kampi, 2024-03-14:
Hi @nullasaservice,

thanks for your request. The main issue with this is how we can integrate the NFC antenna in a propper way and I´m not sure if the MCU can emulate a tag.

closed 2024-03-20 by Kampi

nullasaservice, 2024-03-20:
Hey there! I guess as there has been no response, this has been closed because the MCU does not support it, right? 🥲

Kampi, 2024-03-20:
Hi @nullasaservice 
the MCU does support it but we don't have the hardware for NFC ready yet and we are not sure if we can integrate it properly.