ZSWatch/ZSWatch #257: Implement Chronos App support

jakkra, 2024-04-17:
I created a skeleton here: https://github.com/jakkra/ZSWatch/tree/jakkra_ble_chronos_app

Commands: https://github.com/fbiego/dt78/blob/master/DT78_COMMANDS.md

Find more info in https://github.com/jakkra/ZSWatch/issues/254

fbiego, 2024-04-17:
I have forked the repo and will be working on it soon. Just a quick question, what name and service(s) does the watch advertise for ble discovery?

jakkra, 2024-04-17:
> I have forked the repo and will be working on it soon. Just a quick question, what name and service(s) does the watch advertise for ble discovery?

Just the NUS UUID. If you have a Linux computer you can run everything in Linux (even with Bluetooth) if you want to experiment :) https://github.com/jakkra/ZSWatch/blob/main/GETTING_STARTED.md#1-native-posix 

Although Zephyr can be a bit daunting if you have not worked with it before.

fbiego, 2024-04-17:
Yeah, haven't worked with Zephyr before but I'll try to set it up

fbiego, 2024-05-15:
I haven't had any luck setting up Zephyr. Considering this, I'm contemplating rewriting the Chronos library in C.

jakkra, 2024-05-15:
> I haven't had any luck setting up Zephyr. Considering this, I'm contemplating rewriting the Chronos library in C.

I guess you turned on CPP support in Zephyr?
Of you rewrite it in c, I suggest having a port layer with transmit and receive functions, and then your target platform can implement those however suitable.

fbiego, 2024-06-01:
Hi @jakkra and @Kampi,

I have created PR #290, which still needs review, particularly regarding the implementation of the functions.

Could you please review the code and provide guidance on how to properly implement these functions? I'm currently unable to test on my end as I have not gotten Zephyr running yet.

jakkra, 2024-06-02:
> Hi @jakkra and @Kampi,
>
> I have created PR #290, which still needs review, particularly regarding the implementation of the functions.
>
> Could you please review the code and provide guidance on how to [...]

Ho, very nice, I'm busy now for 2 days,  but will look into it after that!

fbiego, 2025-02-27:
I had a successful test with Chronos app ble integration, I will cleanup the code over the weekend and create a PR

fbiego, 2025-03-29:
Is it possible to use bin files instead of c array on posix? 
Just curious because I want to try custom bin watchfaces saved on external flash.
This would mean watchfaces can be installed without recompiling firmware.

https://www.youtube.com/watch?v=qXx6tj7s6pQ
https://github.com/fbiego/esp32-c3-mini/blob/main/src/ui/custom_face.c