ZSWatch/ZSWatch #272: Issue with wrong hours in new watch faces

Kampi, 2024-05-13:
The hours arent´set correctly. If you have a flip from  i. e. 23 o`clock to 5 o`clock because of a watchface activation at this time points, the leading `2` isn´t deleted which results in 25 o`clock.

My guess is that this issue comes from this line in the generated code:

https://github.com/jakkra/ZSWatch/blob/684d88b40d7297c20686fd16d79102b8d2b129b0/app/src/ui/watchfaces/73_2_dial/zsw_watchface_73_2_dial_ui.c#L338

I think an additional `else` where the image source for the number `0` is loaded should fix the issue.

jakkra, 2024-05-13:
Will check it out

jakkra, 2024-05-13:
Your example seems to work, will debug more 
[Screencast from 2024-05-13 11:34:33.webm](https://github.com/jakkra/ZSWatch/assets/4318648/723454cc-62c8-4547-aa81-dfdbb75a2180)

Kampi, 2024-05-26:
@jakkra I think this is already fixed, isn´t it?

jakkra, 2024-05-26:
> @jakkra I think this is already fixed, isn´t it?

Nope, had it yesterday. Needs debugging

Kampi, 2024-07-23:
Is this still relevant?

jakkra, 2024-07-23:
> Is this still relevant?

Yep not working correctly, needs fix

HongNguyen635, 2025-08-21:
Hello, I recently noticed this issue happened with both the hours and the minutes.
When running on native_sim, there is no issue, but when running actual hardware, it happens quite frequent.
I did a few tests and noticed that it only happens when the display is in an inactive state and the flip happen. Here's what I tried:
- Keep display on, flip from 23:59 -> 00:00, works fine.
- Display inactive, flip from 23:59 -> 00:00 -> wake display up -> time display as 20:50 -> go to a different screen -> go back and time display as correct.

I can continue pick up on this issue, but since I noticed there is some re-organization going on, just want to ask if there's already any work going on with this?

Kampi, 2025-08-22:
You can update the code to LVGL 9 too, if you want because we need it anyway. Please open a PR on the esp32-lvgl-watchface project to avoid splitting it into too many repos.

@jakkra 
Do we have a chance to move my repo into the ZSWatch orga?

jakkra, 2025-08-22:
@HongNguyen635 Awesome find! I'd say fix the already converted watchfaces in this repo, but if you want send a PR with updating here in the code generation also https://github.com/Kampi/esp32-lvgl-watchface/blob/c93644eed7f496435e465ff9be65f94afcf67303/bin2lvgl.kt#L500 you can do that.

There has been some manual modifactions to the generated watchfaces so re-genrating them is not suggested.

The watchfaces work with LVGL 9, so my guess is just naming of the lvgl types that is not LVGL9 (lvgl kept both old and new name for all renamed functions).

@Kampi Imported now, you can delete yours https://github.com/ZSWatch/lvgl-watchfaces
@HongNguyen635 Please submit the PR to this repo instead since it's now moved to ZSWatch org.

closed 2025-08-28 by Kampi