ZSWatch/ZSWatch #286: magnetic dock connector

schotter, 2024-05-30:
Hi, are you using EDAC's [POGO+](https://edac.net/products/spring-loaded-connectors/225)?

jakkra, 2024-05-30:
Nope, it's custom. Or a low profile pin header as alternative.
I have searched everywhere to find a low profile magnetic connector with 5 connections, but impossible.
If you know one I'll happily change :)

schotter, 2024-05-30:
I saw this [post](https://www.mikrocontroller.net/topic/560317#new) months ago and your mechanism reminded me of it.
There exist 5 pin versions of those [POGOs](https://www.digikey.de/en/products/detail/edac-inc/685d05c2p51110e/21299673) but the minimum quantity is 100 (digikey and mouser). By chance I stumbled upon connectors from [Mill](https://www.digikey.de/en/products/detail/mill-max-manufacturing-corp/858-10-005-10-012000/7667983) [Max](https://www.digikey.de/en/products/detail/mill-max-manufacturing-corp/858-22-005-10-011101/7313772), but I have no experience with those... So just for your inspiration ;)

closed 2024-05-30 by schotter

jakkra, 2024-05-30:
> I saw this [post](https://www.mikrocontroller.net/topic/560317#new) months ago and your mechanism reminded me of it. There exist 5 pin versions of those [...]

Thanks, will take a look, maybe something new came in last year, was ast summer I searched all internet for one. Often issue wa they where too thick, want them thin so they don't add too much thickness to the watch.