ZSWatch/ZSWatch #309: Using the RTC update for the 1s periodic event

Kampi, 2024-07-15:
Replace the timer for the 1s periodic callback with the RTC update function when using rev 5

[image]

Kampi, 2024-07-15:
@jakkra maybe we can replace the 10 seconds timer too?

jakkra, 2024-07-15:
@Kampi make sure both 1s and 10s are still executed in the system workqueue context. i.e. switch the delayed work to just a k_work when the RTC ISR happens.

Drawback of using it also for 10s, is that we still need to wake up every 1s, even if there are none listening on 10s. Lower accuracy on 10s is fine IMO.

Kampi, 2024-07-15:
@jakkra good point. We also can't use the alarm for this because we need the alarm for something like a stopwatch. So let's start with 1s first. 
I want to use a worker scheduling in the update callback to make sure that there is no difference. So it's the same scenario for all listeners as now.

Kampi, 2024-07-15:
@jakkra the question how to add this to the code. We can add only one update callback to the driver. So we need a bit of gluecode to handle more than one callback or I move the code from `zsw_periodic_event` to `zsw_clock` but I don´t like this solution because it splits up the code.

jakkra, 2024-07-15:
> @jakkra the question how to add this to the code. We can add only one update callback to the driver. So we need a bit of gluecode to handle more than one callback or I move the code from `zsw_periodic_event` to `zsw_clock` but I don´t like this solution because it splits up the code.

Just from the top of my head how about register the rtc callback in zsw_clock and then add a way to register a tick callback to zsw_clock? Which then `zsw_periodic_event` uses?

Kampi, 2024-07-15:
> Just from the top of my head how about register the rtc callback in zsw_clock and then add a way to register a tick callback to zsw_clock? Which then `zsw_periodic_event` uses?

That´s what I mean with gluecode to add more than one callback :)

jakkra, 2024-07-15:
> That´s what I mean with gluecode to add more than one callback :)

Don't have a better suggestion :/

PR #283 (2024-07-17): Add RV-8263-C8 driver
- Add RV-8263-C8 driver
- Some smaller fixes

Closes #284 
Closes #267 
Closes #285 
Closes #281 
Closes #309
Closes #311
PR #318 (2024-09-12): Add RTC to firmware
Closes #309
closed 2024-09-12 by jakkra