ZSWatch/ZSWatch #329: Rework MCUboot

Kampi, 2024-07-25:


vChavezB, 2024-08-01:
I have seen that the project is now using ncs 2.7.0, which would make it easier to integrate sysbuild instead of partition manager 👍

Kampi, 2024-08-02:
@jakkra do you know sysbuild?

jakkra, 2024-08-02:
> @jakkra do you know sysbuild?

No but this needs to e updated to use. Current child image way is depricated.

> I have seen that the project is now using ncs 2.7.0, which would make it easier to integrate sysbuild instead of partition manager 👍

But sysbuild doesn't not remove partition manager, it's still needed I think? It's different things.

vChavezB, 2024-08-02:
I have been testing compatibility of sysbuild with NCS sdk and so far it works almost as if I was using Zephyr upstream.

In sysbuild you need to specifically disable partition manager in your  `sysbuild.conf`` with option `SB_CONFIG_PARTITION_MANAGER=n`

jakkra, 2024-08-02:
> I have been testing compatibility of sysbuild with NCS sdk and so far it works almost as with Zephyr upstream.
> 
> In sysbuild you need to specifically disable partition manager in your `sysbuild.conf` with option `SB_CONFIG_PARTITION_MANAGER=n`

Ok nice, will have to try that. Getting rid if the partitions.yml and all the partition manager defines would be nice as I think it just removes flexibility for no real gain. I prefer the Zephyr way of defining partitions.

jakkra, 2025-06-15:
Fixed in https://github.com/ZSWatch/ZSWatch/pull/385 and https://github.com/ZSWatch/ZSWatch/pull/388

closed 2025-06-15 by jakkra