ZSWatch/ZSWatch #351: Build  latest Master error

tianmaduxing, 2024-09-08:
Could you help me fix the build error? thanks!

**I have updated the submodule and followed the readme set up the enviroment.**

[image]

Building app
C:\WINDOWS\system32\cmd.exe /d /s /c "west build --build-dir c:/Repos/ZSWatch/app/build c:/Repos/ZSWatch/app --pristine --board zswatch_nrf5340_cpuapp@4 -- -DNCS_TOOLCHAIN_VERSION=NONE -DCONF_FILE=c:/Repos/ZSWatch/app/boards/debug.conf -DOVERLAY_CONFIG=boards/debug.conf -DBOARD_ROOT=c:/repos/zswatch/app;c:/repos/zswatch/zephyr"

-- west build: generating a build system
PCB revision: 4
Loading Zephyr default modules (Zephyr base).
-- Application: C:/Repos/ZSWatch/app
-- CMake version: 3.21.0
-- Found Python3: C:/ncs/toolchains/ce3b5ff664/opt/bin/python.exe (found suitable version "3.9.13", minimum required is "3.8") found components: Interpreter 
-- Cache files will be written to: C:/Repos/ZSWatch/zephyr/.cache
-- Zephyr version: 3.4.99 (C:/Repos/ZSWatch/zephyr)
-- Found west (found suitable version "1.2.0", minimum required is "0.14.0")
-- Board: zswatch_nrf5340_cpuapp, Revision: 4
-- Found host-tools: zephyr 0.16.5 (C:/ncs/toolchains/ce3b5ff664/opt/zephyr-sdk)
-- Found toolchain: zephyr 0.16.5 (C:/ncs/toolchains/ce3b5ff664/opt/zephyr-sdk)
-- Found Dtc: C:/ncs/toolchains/ce3b5ff664/opt/bin/dtc.exe (found suitable version "1.4.7", minimum required is "1.4.6") 
-- Found BOARD.dts: C:/Repos/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp.dts
-- Found devicetree overlay: C:/Repos/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp_4.overlay
-- Generated zephyr.dts: C:/Repos/ZSWatch/app/build/zephyr/zephyr.dts
-- Generated devicetree_generated.h: C:/Repos/ZSWatch/app/build/zephyr/include/generated/devicetree_generated.h
-- Including generated dts.cmake file: C:/Repos/ZSWatch/app/build/zephyr/dts.cmake

warning: BT_DIS_HW_REV_STR (defined at
C:/Repos/ZSWatch/zephyr/subsys/bluetooth/services\Kconfig.dis:128,
C:/Repos/ZSWatch/zephyr/subsys/bluetooth/host\../services\Kconfig.dis:128,
subsys/bluetooth\host\../services\Kconfig.dis:128) was assigned the value '4' but got the value ''.
Check these unsatisfied dependencies: ((BT_DIS_HW_REV && BT_DIS && BT_CONN && BT_RPC_STACK) ||
(BT_DIS_HW_REV && BT_DIS && BT_CONN && BT_CONN && BT_HCI_HOST && BT_RPC_STACK) || (BT_DIS_HW_REV &&
BT_DIS && BT_CONN && BT_CONN && BT_HCI_HOST && BT_HCI && BT)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_BT_DIS_HW_REV_STR and/or look up
BT_DIS_HW_REV_STR in the menuconfig/guiconfig interface. The Application Development Primer, Setting
Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful
too.

warning: PM_PARTITION_SIZE_SETTINGS_STORAGE (defined at
C:/Repos/ZSWatch/nrf\subsys\partition_manager\Kconfig.template.partition_config:2) was assigned the
value '0x100000' but got the value ''. Check these unsatisfied dependencies: (SETTINGS_FCB ||
SETTINGS_NVS) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_PM_PARTITION_SIZE_SETTINGS_STORAGE and/or
look up PM_PARTITION_SIZE_SETTINGS_STORAGE in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

warning: PM_PARTITION_REGION_SETTINGS_STORAGE_EXTERNAL (defined at
C:/Repos/ZSWatch/nrf\subsys\partition_manager\Kconfig.template.partition_region:2) was assigned the
value 'y' but got the value 'n'. Check these unsatisfied dependencies: (SETTINGS_FCB ||
SETTINGS_NVS) (=n). See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_PM_PARTITION_REGION
_SETTINGS_STORAGE_EXTERNAL and/or look up PM_PARTITION_REGION_SETTINGS_STORAGE_EXTERNAL in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: LV_Z_USE_FILESYSTEM (defined at modules/lvgl/Kconfig:33, modules\lvgl\Kconfig:33) was
assigned the value 'y' but got the value 'n'. Check these unsatisfied dependencies: ((FILE_SYSTEM &&
LVGL) || (FILE_SYSTEM && LVGL && 0)) (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_LV_Z_USE_FILESYSTEM and/or look up
LV_Z_USE_FILESYSTEM in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: UART_CONSOLE (defined at drivers/console/Kconfig:43) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: SERIAL_HAS_DRIVER (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_UART_CONSOLE and/or look up UART_CONSOLE in
the menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration
Values, and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: APDS9306 (defined at C:/Repos/ZSWatch/app\drivers\sensor\apds9306/Kconfig:7) was assigned
the value 'y' but got the value 'n'. Check these unsatisfied dependencies: SENSOR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_APDS9306 and/or look up APDS9306 in the
menuconfig/guiconfig interface. The Application Development Primer, Setting Configuration Values,
and Kconfig - Tips and Best Practices sections of the manual might be helpful too.

warning: APDS9306_IS_APDS9306_065 (defined at
C:/Repos/ZSWatch/app\drivers\sensor\apds9306/Kconfig:16) was assigned the value 'y' but got the
value 'n'. Check these unsatisfied dependencies: APDS9306 (=n), SENSOR (=n). See
http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_APDS9306_IS_APDS9306_065 and/or look up
APDS9306_IS_APDS9306_065 in the menuconfig/guiconfig interface. The Application Development Primer,
Setting Configuration Values, and Kconfig - Tips and Best Practices sections of the manual might be
helpful too.

warning: The choice symbol SENSOR_LOG_LEVEL_WRN (defined at
subsys/logging/Kconfig.template.log_config:14) was selected (set =y), but no symbol ended up as the
choice selection. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_SENSOR_LOG_LEVEL_WRN
and/or look up SENSOR_LOG_LEVEL_WRN in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

warning: The choice symbol DEBUG_COREDUMP_BACKEND_OTHER (defined at
subsys/debug\coredump/Kconfig:38) was selected (set =y), but no symbol ended up as the choice
selection. See http://docs.zephyrproject.org/latest/kconfig.html#CONFIG_DEBUG_COREDUMP_BACKEND_OTHER
and/or look up DEBUG_COREDUMP_BACKEND_OTHER in the menuconfig/guiconfig interface. The Application
Development Primer, Setting Configuration Values, and Kconfig - Tips and Best Practices sections of
the manual might be helpful too.

Parsing C:/Repos/ZSWatch/app/Kconfig
Loaded configuration 'C:/Repos/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp_defconfig'
Merged configuration 'C:/Repos/ZSWatch/app/boards/arm/zswatch_nrf5340/zswatch_nrf5340_cpuapp_4.conf'
Merged configuration 'c:/Repos/ZSWatch/app/boards/debug.conf'
Merged configuration 'C:/Repos/ZSWatch/app/boards/debug.conf'
Merged configuration 'C:/Repos/ZSWatch/app/build/zephyr/misc/generated/extra_kconfig_options.conf'
warning: the default selection BT_HCI_IPC (undefined) of <choice BT_HCI_BUS_TYPE> (defined at c:/repos/zswatch/zephyr/boards/shields\x_nucleo_idb05a1\Kconfig.defconfig:11, boards/shields\x_nucleo_idb05a1\Kconfig.defconfig:11, C:/Repos/ZSWatch/app/boards/arm/zswatch_nrf5340/Kconfig.defconfig:76, drivers/bluetooth/hci/Kconfig:11) is not contained in the choice

error: Aborting due to Kconfig warnings

CMake Error at C:/Repos/ZSWatch/zephyr/cmake/modules/kconfig.cmake:348 (message):
  command failed with return code: 1
Call Stack (most recent call first):
  C:/Repos/ZSWatch/nrf/cmake/modules/kconfig.cmake:29 (include)
  C:/Repos/ZSWatch/zephyr/cmake/modules/zephyr_default.cmake:129 (include)
  C:/Repos/ZSWatch/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:66 (include)
  C:/Repos/ZSWatch/zephyr/share/zephyr-package/cmake/ZephyrConfig.cmake:92 (include_boilerplate)
  CMakeLists.txt:45 (find_package)

-- Configuring incomplete, errors occurred!
FATAL ERROR: command exited with status 1: 'C:\ncs\toolchains\ce3b5ff664\opt\bin\cmake.EXE' -DWEST_PYTHON=C:/ncs/toolchains/ce3b5ff664/opt/bin/python.exe '-Bc:\Repos\ZSWatch\app\build' -GNinja -DBOARD=zswatch_nrf5340_cpuapp@4 -DNCS_TOOLCHAIN_VERSION=NONE -DCONF_FILE=c:/Repos/ZSWatch/app/boards/debug.conf -DOVERLAY_CONFIG=boards/debug.conf '-DBOARD_ROOT=c:/repos/zswatch/app;c:/repos/zswatch/zephyr' '-Sc:\Repos\ZSWatch\app'

tianmaduxing, 2024-09-08:
It's better after west update

But now:
C:/Repos/ZSWatch/app/src/ui/watchfaces/73_2_dial/zsw_watchface_73_2_dial_ui.c:9:10: fatal error: lvgl.h: No such file or directory
    9 | #include <lvgl.h>

jakkra, 2024-09-08:
> It's better after west update
> 
> But now: C:/Repos/ZSWatch/app/src/ui/watchfaces/73_2_dial/zsw_watchface_73_2_dial_ui.c:9:10: fatal error: lvgl.h: No such file or directory 9 | #include <lvgl.h>

Make sure you do a pristine build as you built once without west update correctly done. Also please share full log. It's compiling on ci so it's probably something with your setup that's wrong.

closed 2024-10-09 by jakkra