ZSWatch/ZSWatch #361: Prototype Fund

blubbsy, 2024-10-17:
Hi,

the german open knowledge fundation started togetther with the German Government again a fund for open source projects: https://hardware.prototypefund.de/

it funds 10 projects with 24.500€ for one year and after 12 month a presentable prototype should be done. Is it something ZSWatch should try to apply for? I also would understand that you wont like it, because it somehow builds up pressure to deliver something, but on the other side the big amount of time you invested would have some return :) 

Though they write it's focus is on environmental topics the examples show laser cutters, pen holders and more: 
https://hardware.prototypefund.de/en/projects-2/

jakkra, 2024-10-17:
Hi, thanks for the suggestion! I have already applied at https://nlnet.nl/funding.Html and should get reply soon. But will look this up if I don't get that!

jakkra, 2024-10-17:
> For the German fund it is time until the end of the month. Why not apply to both and in the worst case you reject one if them? Hope that it works, but I can check the participation conditions.

I see, I'll dig into it a bit. Can probably reuse alot from the other application, thanks :)

jakkra, 2024-10-17:
Oh, I can't apply. But @Kampi if you feel for it, you could give it a try. 

> Due to our jury and funding regulations the application has to be in German and has to be submitted by someone with residency in Germany.

Kampi, 2024-10-18:
Yeah I can do it over the weekend.

Kampi, 2024-10-22:
We prepared the first draft for the fund. I will tune it a bit until the end of the month and then we have to cross the fingers :)

blubbsy, 2024-10-24:
Do you want me to take a look on it or are you ok?

Kampi, 2024-10-24:
@blubbsy sure. Can you share your e-mail address with me, so I can invite you?

blubbsy, 2024-10-24:
both possible ;) 

discord username: devsibwarra

closed 2024-10-24 by Kampi