ZSWatch/ZSWatch #375: gc9a01 driver screen cannot be displayed

fghill, 2025-03-22:
I encountered a problem where the screen failed to display when developing the screen display using Zephyr. My screen is a 1.28-inch circular color TFT display with a high-definition IPS module of 240*240 and an SPI interface, driven by GC9A01. I called the driver file display_gc9x01x.c under the directory F:\zephyr\zephyrproject\zephyr\drivers\display. When I ran "west build -p always -b esp32s3_devkitm", there were no error reports. However, after burning it into my mainboard, the display function failed to work, even showing a black screen. Is it a compatibility issue with my driver?

jakkra, 2025-03-22:
Are you running ZSWatch on an ESP32 board? ZSWatch use a modified Gc9a01 driver in order to o support low power modes better. In theory I think using the Zephyr GC9A01 driver should work, as it will just always be on. But don't see why not use ZSWatch driver as it is in the repo and built automically.

fghill, 2025-03-23:
Thank you for your reply. I used esp32s3-n8r2 for development. I defined the SDA, SCL, CS, RES and DC of the screen in the pin under gpio1 of esp32.overlay file, that is, gpio38 to gpio42. I used the driver in the zephyr driver directory. Now I am ready to use the driver you provided. If I succeed, I will come back and tell you.

closed 2025-03-23 by jakkra