ZSWatch/ZSWatch #377: nrf5340dk/nfr5340/cpuapp compile error

HiJackhem, 2025-03-27:
Hello，
zswatch@5/nrf5340/cpuapp compiles successfully.

when i change board to  nrf5340dk/nfr5340/cpuapp, it fails.

The error log is as below:

In file included from E:/Projects/InnoPhase/watch/software/ZSWatch/zephyr/include/zephyr/storage/flash_map.h:277,
                 from E:/Projects/InnoPhase/watch/software/ZSWatch/app/src/filesystem/zsw_rtt_flash_loader.c:28:
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:43:22: error: 'PM_lvgl_raw_partition_ID' undeclared here (not in a function)
   43 | #define PM_ID(label) PM_##label##_ID
      |                      ^~~
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:48:35: note: in expansion of macro 'PM_ID'
   48 | #define FIXED_PARTITION_ID(label) PM_ID(label)
      |                                   ^~~~~
E:/Projects/InnoPhase/watch/software/ZSWatch/app/src/filesystem/zsw_rtt_flash_loader.c:62:6: note: in expansion of macro 'FIXED_PARTITION_ID'
   62 |     [FIXED_PARTITION_ID(lvgl_raw_partition)] = "lvgl_raw_partition",
      |      ^~~~~~~~~~~~~~~~~~
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:43:22: error: array index in initializer not of integer type
   43 | #define PM_ID(label) PM_##label##_ID
      |                      ^~~
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:48:35: note: in expansion of macro 'PM_ID'
   48 | #define FIXED_PARTITION_ID(label) PM_ID(label)
      |                                   ^~~~~
E:/Projects/InnoPhase/watch/software/ZSWatch/app/src/filesystem/zsw_rtt_flash_loader.c:62:6: note: in expansion of macro 'FIXED_PARTITION_ID'
   62 |     [FIXED_PARTITION_ID(lvgl_raw_partition)] = "lvgl_raw_partition",
      |      ^~~~~~~~~~~~~~~~~~
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:43:22: note: (near initialization for 'partition_map')
   43 | #define PM_ID(label) PM_##label##_ID
      |                      ^~~
E:/Projects/InnoPhase/watch/software/ZSWatch/nrf/include/flash_map_pm.h:48:35: note: in expansion of macro 'PM_ID'
   48 | #define FIXED_PARTITION_ID(label) PM_ID(label)
      |                                   ^~~~~
E:/Projects/InnoPhase/watch/software/ZSWatch/app/src/filesystem/zsw_rtt_flash_loader.c:62:6: note: in expansion of macro 'FIXED_PARTITION_ID'
   62 |     [FIXED_PARTITION_ID(lvgl_raw_partition)] = "lvgl_raw_partition",
      |      ^~~~~~~~~~~~~~~~~~
[806/1338] Building C object CMakeFiles/app.dir/src/filesystem/zsw_lvgl_spi_decoder.c.obj

jakkra, 2025-03-27:
Hi, need more info. But my guess is that you are building with sysbuild.
Add `--no-sysbuild` if you builf from command line, or check this box if you use vscode

[image]

jakkra, 2025-03-27:
You can also check the ci build for the nrf5340dk and compary the arguments to see what you miss:
https://github.com/ZSWatch/ZSWatch/actions/runs/14036309420/job/39295046912

HiJackhem, 2025-03-27:
Hi

Thank you for your response.

[image]

closed 2025-03-27 by HiJackhem

jakkra, 2025-03-27:
Did it work?

HiJackhem, 2025-03-27:
I will check the ci build for the nrf5340dk and see whether something miss

jakkra, 2025-03-27:
Ok, it doesn't work for me either. It's something Nordic added/changed in their VSCode plugin that cause this issue. Building from command line work for me. Let me check a bit.

HiJackhem, 2025-03-27:
Does the command line work on windows? I build it on windows vscode.

jakkra, 2025-03-27:
> Does the command line work on windows? I build it on windows vscode.

I have never built it on Windows, neither in vscode or teminal. But if one works, the other will also.

Instead of finding out why vscode doesn't work, just use this branch: jakkra_hw_model_v2_migrate
I updated to Zephyr HW Model 2 for the project in this branch and that fixes the incompatability with VSCode plugin.

But I did not verify this on nordic DK if everything works, but it builds at least.

https://github.com/ZSWatch/ZSWatch/tree/jakkra_hw_model_v2_migrate

HiJackhem, 2025-03-27:
I checkout the branch jakkra_hw_model_v2_migrate and it works now. Thank you very much

jakkra, 2025-03-27:
> I checkout the branch jakkra_hw_model_v2_migrate and it works now. Thank you very much

I'll merge this to main soon. Only reason I have not yet is that for some reason v4 of ZSWatch then display won't turn on and have not prioritized to fix it.