ZSWatch/ZSWatch #381: error realtime display on watchface

HiJackhem, 2025-04-08:
Start time should be 1970-01-01, not 1970-00-01. It will lead to the real month less one.

[image]

[image]

jakkra, 2025-04-08:
Hehe you are right, can see it myself now. Will fix once I get a chance. Feel free to submit a PR if you know where to change.

Kampi, 2025-04-08:
Do you use the external RTC? There was a bug in the RTC driver which I have fixed in Upstream Zephyr:

https://github.com/zephyrproject-rtos/zephyr/commit/db9b45a7b3f0d2a2b63a840d8d7d3abfcdb7709b

Maybe this fix isn´t ported to ZSWatch currently?

jakkra, 2025-04-08:
> Do you use the external RTC? There was a bug in the RTC driver which I have fixed in Upstream Zephyr:
> 
> [zephyrproject-rtos/zephyr@db9b45a](https://github.com/zephyrproject-rtos/zephyr/commit/db9b45a7b3f0d2a2b63a840d8d7d3abfcdb7709b)
> 
> Maybe this fix isn´t ported to ZSWatch currently?

I got the same issue without RTC myself. So not related.

PR #382 (2025-04-09): Fix most watchfaces showing month one off. 
Fixes #381