ZSWatch/ZSWatch #393: Buying device

sx1262-test, 2025-06-24:
Hi Jakob,

More by incident than really looking for this I saw a video about your watch. First of all, respect for this cool work. 
Is it possible to get a ZSWatch or to buy it somewhere? I'm really interested in trying it out and write code for it to expand it for my purposes if not covering all already.

Best

Andreas

jakkra, 2025-06-24:
Hi, we have developed a development kit we currently are waiting to receive to validate the design. Once that's done we will have some of those to sell for people who want to get started early. Suggest you read here and sign up on the mail list where we will keep you updated about progress. https://zswatch.dev/blog

sx1262-test, 2025-06-25:
Hi, many thanks. I put my address on the list. Andreas