ZSWatch/ZSWatch #406: Issue with Bluetooth Connection on iOS

HongNguyen635, 2025-08-11:
I'm current on iOS 18.5, and is using nRF Connect Mobile app to connect.
I'm on debug mode, and the log shows that the connection is successful and the Bluetooth icon appeaers on the screen as well, however, time is not updated on the watch. I tried to switch to another phone also on iOS 18 and the same issue persists.

Below is a snipet from the log:
```
[00:00:11.833,312] <inf> ble_comm: Connected <the address> (random)
[00:00:11.833,496] <inf> ble_comm: MTU exchange pending
[00:00:11.833,526] <inf> ble_comm: Interval: 24, latency: 0, timeout: 72
[00:00:11.833,557] <inf> ble_chronos: Bluetooth  Connected
[00:00:11.836,578] <wrn> bt_hci_core: opcode 0x2022 status 0x0c 
[00:00:11.836,608] <err> bt_conn: Failed to set data len (-13)
[00:00:12.094,451] <inf> ble_comm: LE PHY updated: TX PHY LE 2M, RX PHY LE 2M
[00:00:12.154,724] <inf> ble_comm: LE PHY updated: TX PHY LE 2M, RX PHY LE 2M
[00:00:12.184,356] <inf> ble_comm: LE data len updated: TX (len: 251 time: 2120) RX (len: 251 time: 2120)

[00:00:12.184,783] <inf> ble_comm: MTU exchange done. 244
[00:00:12.185,394] <wrn> bt_l2cap: Ignoring data for unknown channel ID 0x003a
[00:00:12.425,292] <dbg> ble_cts: discover_completed_cb: The discovery procedure succeeded```

HongNguyen635, 2025-08-11:
I'm building with RTC, and with the head up-to-date with main branch.

jakkra, 2025-08-11:
> I'm building with RTC, and with the head up-to-date with main branch.

You could try without, there is a conf no_rtc.conf to build with, don't think that's the problem, but you could try if you want. Anyway I'll check tonight when I have access to an iPhone.

HongNguyen635, 2025-08-12:
I'm closing this issue because it's been addressed in #407